rust-embedded/embedded-hal #608: Shared SPI bus between devices with different SPI modes

dan-corneanu, 2024-06-16:
I have a small board that has 2 devices attached to the same SPI bus
* AD9833 DDS that uses SPI MODE 2
* MCP41010 Digital Potentiometer that can use SPI MODE 0 or MODE 3

Each device has its own CS pin.

I am using the following two drivers
* https://github.com/eldruin/ad983x-rs.git
* https://github.com/dan-corneanu/mcp4x-rs.git

The problem that I am facing is that my program can only communicate with one of the devices because
the `SpiDevice` can not change the mode of the `SpiBus`.

I think this design decision was discussed in 2022 in this comment https://github.com/rust-embedded/embedded-hal/pull/351#issuecomment-1043686888.

Could you please advise on how I could address this problem in my code?

If interested in the code of my application you can have a look at this branch https://github.com/dan-corneanu/ad983x_signal_generator/tree/features/configure_ushell.

eldruin, 2024-06-21:
At the moment this must be implemented in a HAL-specific way. The difficulty will depend on the HAL you are using (I cannot access your application repo) but it will probably involve rewriting a bit of `embedded-hal-bus` so that when a device is activated, the bus is reconfigured.

dan-corneanu, 2024-06-21:
Have made it public.