rust-embedded/embedded-hal #610: The Crate Does not Compile when the `defmt-03` Feature is Enabled

joboudreault, 2024-06-27:
## Problem

Currently (on master commit `1c585d4`), the crate `embedded-hal` and `embedded-io` does not compile when enabling the `defmt-03` feature. The following command is run : 

```
[./embedded-hal] $ cargo build --features defmt-03;
```

And these errors are returned :

<details> 
  <summary>Errors</summary>

```
error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/i2c.rs:192:41
    |
192 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
164 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/i2c.rs:192:41
    |
192 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/i2c.rs:304:41
    |
304 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
164 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/i2c.rs:304:41
    |
304 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/spi.rs:201:41
    |
201 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
174 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/spi.rs:201:41
    |
201 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/spi.rs:319:41
    |
319 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this crate through its public re-export
    |
174 + use crate::defmt;
    |

error[E0433]: failed to resolve: could not find `defmt` in the list of imported crates
   --> embedded-hal/src/spi.rs:319:41
    |
319 | #[cfg_attr(feature = "defmt-03", derive(defmt::Format))]
    |                                         ^^^^^^^^^^^^^ could not find `defmt` in the list of imported crates
    |
    = note: this error originates in the derive macro `defmt::Format` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0433`.
error: could not compile `embedded-hal` (lib) due to 8 previous errors
```
</details>

This is run by the following Rust version : 
```
cargo 1.79.0 (ffa9cf99a 2024-06-03)
release: 1.79.0
commit-hash: ffa9cf99a594e59032757403d4c780b46dc2c43a
commit-date: 2024-06-03
host: x86_64-unknown-linux-gnu
libgit2: 1.7.2 (sys:0.18.3 vendored)
libcurl: 8.6.0-DEV (sys:0.4.72+curl-8.6.0 vendored ssl:OpenSSL/1.1.1w)
ssl: OpenSSL 1.1.1w  11 Sep 2023
os: Arch Linux [64-bit]
```

## Solution
I forked the repository and pushed [a branch](https://github.com/joboudreault/embedded-hal/tree/610/fix-defmt-03-build) that successfully compiles `embedded-hal` and `embedded-io` in a standalone fashion. It is confirmed to work for our needs. 

However, I know this could conflict with a previous discussion on PR https://github.com/rust-embedded/embedded-hal/pull/481.

What is happening in this case ?

Thanks for your help.

Dirbaio, 2024-06-27:
I cannot reproduce this

```
git[dirbaio@mercury ~]$ git clone https://github.com/rust-embedded/embedded-hal
Cloning into 'embedded-hal'...
remote: Enumerating objects: 4476, done.
remote: Counting objects: 100% (946/946), done.
remote: Compressing objects: 100% (221/221), done.
remote: Total 4476 (delta 781), reused 781 (delta 725), pack-reused 3530
Receiving objects: 100% (4476/4476), 1.49 MiB | 6.91 MiB/s, done.
Resolving deltas: 100% (2898/2898), done.
[dirbaio@mercury ~]$ cd embedded-hal
[dirbaio@mercury embedded-hal]$ cargo build --features defmt-03
(...)
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 10.05s
[dirbaio@mercury embedded-hal]$ cd embedded-hal
[dirbaio@mercury embedded-hal]$ cargo build --features defmt-03
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.02s
[dirbaio@mercury embedded-hal]$ 
[dirbaio@mercury embedded-hal]$ cargo --version -v
cargo 1.79.0 (ffa9cf99a 2024-06-03)
release: 1.79.0
commit-hash: ffa9cf99a594e59032757403d4c780b46dc2c43a
commit-date: 2024-06-03
host: x86_64-unknown-linux-gnu
libgit2: 1.7.2 (sys:0.18.3 vendored)
libcurl: 8.6.0-DEV (sys:0.4.72+curl-8.6.0 vendored ssl:OpenSSL/1.1.1w)
ssl: OpenSSL 1.1.1w  11 Sep 2023
os: Arch Linux [64-bit]
[dirbaio@mercury embedded-hal]$ rustc --version -v
rustc 1.79.0 (129f3b996 2024-06-10)
binary: rustc
commit-hash: 129f3b9964af4d4a709d1383930ade12dfe7c081
commit-date: 2024-06-10
host: x86_64-unknown-linux-gnu
release: 1.79.0
LLVM version: 18.1.7
```

Perhaps you have an old `Cargo.lock` file somewhere, perhaps there's something wrong with your Rust installation so it's actually using an older rustc or something.

Dirbaio, 2024-06-28:
Same, it build fine for me. Both with and without the `[patch.crates-io]`.

joboudreault, 2024-07-02:
Ok, then I will create a clean Docker container and try reproducing it there. Thanks for your time.