rust-embedded/embedded-hal #611: honor `ErrorKind::Interrupted` in `write_all()`?

rmja, 2024-07-04:
Currently in `write_all()` (https://github.com/rust-embedded/embedded-hal/blob/master/embedded-io-async/src/lib.rs#L134) we do not retry if the return error has `kind() == ErrorKind::Interrupted`. This is different than `std`: https://doc.rust-lang.org/src/std/io/mod.rs.html#1696-1708.

Should `ErrorKind::Interrupted` lead to a silent retry in `write_all()`?

eldruin, 2024-07-04:
Interesting question. However, I would ask then how often should it retry. Either we create an endless loop or we define an arbitrary number which will be difficult to agree upon.
Given the general character or e-io-async, I think the caller is better positioned to put a loop on the call in this case.

Dirbaio, 2024-07-04:
`ErrorKind::Interrupted` is annoying unix legacy-ish. I don't think anyone uses it intentionally anymore, especially now that async exists? it doesn't really exist on embedded, and i'd say it shouldn't exist.

So perhaps another option is to make `embedded_io_adapters::FromStd` do the retry *on the `write()` method*, so we keep `Interrupted` errors to the boundary of embedded-io instead of letting them in?

rmja, 2024-07-04:
That sounds like a really good idea. Should we then also remove `ErrorKind::Interrupted`?

Dirbaio, 2024-07-04:
probably, yes...