rust-embedded/embedded-hal #624: How to read incoming data from SPI MOSI Lane? (not MISO)

Syphixs, 2024-08-17:
I am working on a specific epd which requires to read data from the MOSI line after sending a command but switching the CS pin between each byte. It's unclear to me if this is even achievable with the SPI Bus implementation. If so some guidance would be appreciated. 
Some relevant parts from the documentation:
[image]

Here it also states that a CS pulse is necessary between each read.  
[image]

I thought something like this would be working:
```rust
    pub fn read_otp(&mut self, buffer: &mut [u8]) -> Result<(), E> {
        self.bus.lock(|bus| {
            let mut bus = bus.borrow_mut();

            self.cs.set_low().ok();
            // Command mode
            self.dc.set_low().ok();

            // Send read command
            bus.write(&[0xA2])?;

            self.dc.set_high().ok(); // Data mode

            // Read data
            for byte in buffer.iter_mut() {
                self.cs.set_high().ok();
                self.cs.set_low().ok();

                // Send dummy byte and read the result
                let mut read_buf = [0u8];
                bus.transfer(&mut read_buf, &[0x00])?;
                *byte = read_buf[0];
            }

            self.cs.set_high().ok();
            Ok(())
        })
    }
```

but then i found out that the transfer also wants to read on the MISO line.

jannic, 2024-09-17:
This looks like a non-standard variation of SPI.

While it may (or may not) be supported by the SPI hardware of your microcontroller, it's not something supported by the abstraction layer provided by embedded-hal.

Dominaezzz, 2025-01-29:
Sounds like 3 wire half duplex. e-hal supports it but you have to configure your SPI hardware to operate in this mode.
This issue should be moved to your hal's repo.