rust-embedded/embedded-hal #636: Wrong implementation of `read_exact`

XuJiandong, 2024-10-10:
https://github.com/rust-embedded/embedded-hal/blob/dd0bbbaa5c926628ff03dfd78fb54162fec05ab9/embedded-io/src/lib.rs#L343

The `buf` returned is always empty.

jannic, 2024-10-10:
@XuJiandong: Are you sure?
The code is a little bit tricky, but it should work. Here is the code with some comments added:
```
    fn read_exact(&mut self, mut buf: &mut [u8]) -> Result<(), ReadExactError<Self::Error>> {
        while !buf.is_empty() { // This doesn't actually check if `buf` already contains data, but if it's length is non-zero
            match self.read(buf) {
                Ok(0) => break,
                Ok(n) => buf = &mut buf[n..], // When some data was read, the reference `buf` gets updated to point to the remaining capacity. Not that this does not affect the caller! The code that called `read_exact(&mut buf)` will still see the full size when this method returns.
                Err(e) => return Err(ReadExactError::Other(e)),
            }
        }
        if buf.is_empty() {
            Ok(()) // Now the local `buf` has length zero, ie. no remaining capacity. That means that the original `buf` this method got passed is now full with received bytes.
        } else {
            Err(ReadExactError::UnexpectedEof)
        }
    }
```

XuJiandong, 2024-10-11:
It does work, indeed. However, its semantics differ from the original implementation of read_exact: https://doc.rust-lang.org/src/std/io/mod.rs.html#961. Note that the method's signature is `fn read_exact(&mut self, buf: &mut [u8])`.

jannic, 2024-10-11:
What's different with the method's signature? Are you talking about the `mut` before `buf` in `mut buf: &mut [u8]`?
That's not really part of the method's signature. It's syntactic sugar. Note that these two snippets are equivalent:
```
    fn read_exact(&mut self, mut buf: &mut [u8]) -> Result<(), ReadExactError<Self::Error>> {
```
```
    fn read_exact(&mut self, buf: &mut [u8]) -> Result<(), ReadExactError<Self::Error>> {
        let mut buf = buf;
```
(Here's a playground link to illustrate that: https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=9e707bce8d27716c84d1d1f5f3fc83a9)

This is also reflected in the docs. The only difference in the signatures shown at https://docs.rs/embedded-io/latest/embedded_io/trait.Read.html#method.read_exact and https://doc.rust-lang.org/stable/std/io/trait.Read.html#method.read_exact is the error type returned.

closed 2024-10-11 by XuJiandong