rust-embedded/embedded-hal #670: [DISCUSSION] GPIO Invert API

felipebalbi, 2025-05-22:
### Introduction

Some GPIO controllers have the ability to invert the logic level in HW. This is useful in cases where some generic code samples the state of a pin and take actions depending on the state. Rather than having the generic know about `ActiveHigh` vs `ActiveLow`, the HW invert the logic level and the generic code can rely solely on `is_low()`/`is_high()` as if the level was not inverted.

### Proposal

There are two options here.

1. Add a new `set_inverted(&mut self)` API to the existing `InputPin` struct
2. Add a new `InvertedInputPin` which assumes GPIO is inverted.

### Challenges

- It's unlikely that every GPIO controller is capable of inverting the logic level in HW. How can we guarantee that the HAL's SW implementation is sound?
- Adding a single new function to `InputPin` would allow for drivers to implement that API and maintain all their types. Existing code using `InputPin` would continue to work, but it would be a breaking change if the new method is required.
- Adding a new `InvertedInputPin` would avoid a breaking change, but any piece of code polymorphic over `InputPin` would have to be modified in order to use `InvertedInputPin`.

Dirbaio, 2025-05-29:
The thing is the GPIO traits are explicitly for GPIO, ie they get/set the voltage on a digital pin.  They're not "logical active/inactive signal input". 

Adding support for inversion (in e-h or in a wrapper written by the user) would be "lying" to the driver. The driver would read "voltage is low" while the reality is it's high.

I think if we want to support logical signals it'd make more sense to have `LogicalInput`, `LogicalOutput` traits. Then we could have wrappers that convert from "gpio input" to "logical input" that can be configured to handle the "active low" and "active high" cases (ie invert or not invert). This way the drivers would declare if they expect physical or logical signals, and no lying would be required.

However, I think doing this would be a bit too over-abstracted. It will be confusing for the users, as the logical/physical distinction is somewhat subtle. I'm not sure if it's worth it.

sirhcel, 2025-05-30:
> Simple example: something like [rmk](https://github.com/HaoboGu/rmk). That is, you have a GPIO matrix and you want to scan it. The matrix could, potentially, be wired up such that pressing a key pulls the input GPIO either high or low; rmk could be given inverted GPIOs where applicable, then the scanner remains the same regardless of how the GPIOs were wired up. That being said, this is a configuration detail that could be handled before giving ownership of the pins to rmk.

> IMO it's simpler to keep the traits for phyiscal voltage only, and let drivers that want inversion handle it themselves.

I'm with @Dirbaio and @Wassasin in the sense that handling inversion directly in the GPIO traits likely will cause more confusion than having the scope limited to the physical GPIO pin.

But what about an adapter @felipebalbi? There is @Wassasin's suggestion `InvertedInputPin` adapting from `InputPin` to `InputPin` which expresses the inversion also in the pin's type. There is also the proof-of-concept [switch-hal](https://crates.io/crates/switch-hal) around which seems to have reached some adoption and adapts from `InputPin` to `InputSwitch` clearly separating the concerns of GPIO pin state of the state of some other coupled net. And in https://github.com/rubberduck203/switch-hal/pull/12 there were at least two hands raised to get this closer to to embedded-hal.

Dirbaio, 2025-05-30:
> InvertedInputPin adapting from InputPin to InputPin

this is the "lying" that I think is bad.

`InputSwitch` should take `(InputPin, enum{ ActiveHigh, ActiveLow}` and handle the inversion itself. It should not take just `InputPin` and expect the user to wrap the input pin into an `InvertedInputPin`

sirhcel, 2025-05-30:
> > InvertedInputPin adapting from InputPin to InputPin
> 
> this is the "lying" that I think is bad.

Now I see what you were referencing to. Yes, this would water the scope of a physical GPIO pin down. But if one _really really_ want to do this for something expecting `InputPin` or `OutputPin`, then the adapter approach puts at least some "warning tape" in place. I wanted to state that from my point of view this is somewhat less confusing than making inversion run-time configurable and generally available.

> For example, `InputSwitch` should take `InputPin, enum{ ActiveHigh, ActiveLow}` and handle the inversion itself. It should not take just `InputPin` and expect the user to wrap the input pin into an `InvertedInputPin`
> 
> Similarly for any other driver that wants to take input pins that might be active-high or active-low.

That's what switch-hal's [`Switch`](https://docs.rs/switch-hal/0.4.0/switch_hal/struct.Switch.html) is doing. Not by means of an enum but with a marker type.